irekzielinski/Pylontech-Battery-Monitoring #1: Response is only ????

TMHEINE, 2019-10-15:
Hi,
I did all the same as you did, with the same elements (Wemos D1 mini, Sparkfun Max3232).
I can access the web interface. But all responses I get is always "????".
What is wrong?
[image]
[image]

djsyl, 2019-12-05:
hello
same thing for me !

TMHEINE, 2019-12-09:
I use now the can Port to read the SOC.
RPI3 with PiCan2

djsyl, 2019-12-12:
Now it work for me 
when use a other max3232 not work
with the same of description it's good now

TMHEINE, 2021-04-22:
Hello,

I tried it again all new.
Parts: 
Wemos D1 Mini v3.0.0
Sparkfun Max3232 (Bought a new one

I get the same Questionsmarks as before

Mainpage:
[image]
Power response:
[image]
The log is:
[image]

What could be the Problem? Wireing? Arduino?

Compiling Result Arduino:
[image]

Anyone who can help me?

irekzielinski, 2021-08-05:
@TMHEINE did you get it working?
Your log indicates that ESP is not establising any connection with the battery.
I suspect this is electrical issue/wiring.

TMHEINE, 2021-08-06:
Hello.
Didi not get it working. Electrical/wiring should not be the problem.
Maybe port is not working or something else.
I did not try it again after last test.
For the last test I used a new ESP and new Sparkfun. 
I tried lot of different ways likt exchange RX and TX, Other capasitors, different power supplies.
but nothing worked.

So actually i am reading the batteries by can-hat and python.

Wibet10, 2022-02-12:
Hi,
I built one of these monitors a coupe of years ago and  it worked fine with my UC3000 batteries as described in the original design using the RJ11 port. End of last year I fited a new UC3000C battery as master and had to convert to the 8 pin 'Console' port. I rewired the cable and it started working but was erratic. It then refused to read at all and I abandonned it until this month. On investigation, the RS232 signals were barely 8-10volts pk/pk and the MQTT was trying to reconnect every cycle of the program. I also discovered from the web that the UC2000/3000C batteries do not need the 1200/115200 codes and  rate change to wake them, they are ready to communicate at 115200 on startup. 
I have commented out 'void wakeUpConsole and void switchBaud' sections of the code and added this into the void loop: -
   void loop() {
  // millis delay check below runs 'mqttLoop' every 30 seconds
      Client.loop();
        if (millis() - lastMillis >= 30*1000UL) // >= 30 sec period
    {   
     lastMillis = millis();  // reset ready for next loop
        mqttLoop(); 
    }
   ArduinoOTA.handle();
  server.handleClient();
  timer.run();_
code from this point is original --
The erratic connection problem was fixed by including a level shifter between the ESP and the MAX3232 as in the attached diagram.
This had the effect of increasing the RS232 signal voltages sent to the battery to around 12/15volts pk/pk, much more in spec and very solid connection.
There seems to have been some changes that accomodate the new battry types since I last checked in (November 2021) and reading the above comments, the discoveries I made in adapting my original build may help resove some of the problems experienced by others.
[image]
Hope this helps others connecting to UC2000/3000C batteries.

irekzielinski, 2022-02-12:
Thanks - that some good findings!
Do you really need to wrap mqttLoop(); into this 1in30 seconds condition?
There should be no problem calling this every time loop is executed.

Wibet10, 2022-02-12:
Hi, Thanks for the response. I hope the info. on getting a better RS232 connection helps others.
In my original build I tried several MAX3232's before I found a working one. On testing the voltages on my worker during the upgrade, @3.3v the o/p is minimum spec, @5v it is comfortably in spec.
My coding is 'cut and paste' at best! My original problem was the mqtt disconnecting/reconnecting on every loop. In tracing this back to my mqtt user/password set-up, I broke your mqttloop code (probably the syntax somewhere). My crude fix was to put the timing loop into the main loop and only run the mqttloop every 30 secs. My code now compiles and runs ok, think I'll leave it running for a while and look at fixing it later.
Thanks for a great project that has kept up with the battery developments, cheaper than setting up Batteryview and almost as good, nice one!

Wibet10, 2022-02-16:
Sorry you are still having proiblems. I have mine running using the enclosed code, it compiles using Arduino IDE 1.8.13 & 2.0.0rc3 on a Win10 PC. I have a genuine Wemos Mini Pro and a 'cheap generic' Max3232. I started out in February with no reads from the Pylon-stack (Master is now UC3000C - new type) and only ??? when I accessed the local 192.168.1.71 site. 
I 'fiddled about' with the code and put in a 'lot of debug comments' to get my head around how it worked, I did get it going but I think I broke some of the original code? My changes are around lines 745-798 and 848-870 in the attached version. There are a lot of serial.prints/notes I have put in to help me remember what I did, hope it helps you sort yours. 
DON'T leave any 'extra' Serial.prints in the code or IT WILL totally disrupt the reading of the Pylontech's.
(this is the code as it is working for me now -  copy + paste it into Arduino IDE) 
All credit goes to Irek for his original code.

// WiFi code for 'D1 mini Pro' to run Pylontech Console Port into Node Red server on 192.168.1.197
// Has pylontech battery web server access @ 192.168.1.71 (local wifi port)
// Updated 07-02-2022 Taking out start at 1200 and handshake to then switch to 115200 (older batteries) - now starts at 115200 and just reads.

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#include <ESP8266WebServer.h>
#include <PubSubClient.h>
#include <SimpleTimer.h>
#include <TimeLib.h>        //https://github.com/PaulStoffregen/Time
#include <ntp_time.h>       // this library needs loading manually
#include <circular_log.h>   // this library needs loading manually

//#define DEBUG  // Comment out to remove DEBUG Serial prints - useful to see the loops running
                // if left enabled while connected to pylon - it WILL corrupt the real pylon data reads!!

//IMPORTANT: Specify your WIFI settings:
#define WIFI_SSID "xxxxxxxxxxxxxxxxx"
#define WIFI_PASS "xxxxxxxxxxxxxx"

//IMPORTANT: Uncomment this line if you want to enable MQTT (and fill correct MQTT_ values below):
#define ENABLE_MQTT

#ifdef ENABLE_MQTT
//NOTE 1: if you want to change what is pushed via MQTT - edit function: pushBatteryDataToMqtt. (line 822)
//NOTE 2: MQTT_TOPIC_ROOT is where battery will push MQTT topics. For example "soc" will be pushed to: "pylon_battery/soc"
//Note 3: Using WiFi into Raspberry Pi
#define MQTT_SERVER        "192.168.1.197"  //is Node Red Home Server IP {Home mqtt server 127.0.0.1:1883}
#define MQTT_PORT          1883
#define MQTT_USER          "emonpi"
#define MQTT_PASSWORD      "xxxxxxxxxxxxxx"  
#define MQTT_TOPIC_ROOT    "emon/pylon_battery/"  //this is where mqtt data will be pushed
//#define MQTT_PUSH_FREQ_SEC 30  // MQTT push now set by millis delay in 'void loop'
#endif //for #def ENABLE_MQTT

#define ABS_DIFF(a, b) (a > b ? a-b : b-a)

WiFiClient batClient;
PubSubClient Client(batClient);  // set-up MQTT client

char g_szRecvBuff[7000];
static unsigned long g_lastDataSent = 0; //sets initial MQTT last-sent time for first loop
unsigned long lastMillis;

ESP8266WebServer server(80);
SimpleTimer timer;
circular_log<7000> g_log;
bool ntpTimeReceived = false;
int g_baudRate = 0;

void Log(const char* msg)
{
  g_log.Log(msg);
  #ifdef DEBUG
  Serial.println("logging messages");
  #endif
}
//  'callback' Not needed for Pylon??
void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
[... 801 lines cut ...]
     (sendCommandAndReadSerialResponse("pwr") == true);  }
       
     #ifdef DEBUG
      Serial.println("sendCommandAndReadSerialResponse = PWR == TRUE....");
     #endif 
    {
    static batteryStack lastSentData; //this is the last state we sent to MQTT, used to prevent sending the same data over and over again
    static unsigned int callCnt = 0;
    parsePwrResponse(g_szRecvBuff);

    bool forceUpdate = (callCnt % 6 == 0); //push all the data every 20th call (change to 6 for testing)
    pushBatteryDataToMqtt(lastSentData, forceUpdate);
         // Serial.println("Forcing Data Update ...");
    callCnt++;
         // Serial.println(callCnt++);
    memcpy(&lastSentData, &g_stack, sizeof(batteryStack));
    }
  #ifdef DEBUG
  Serial.println("END of mqtt.loop");
  #endif 
  }

//#endif ENABLE_MQTT

########## END OF CODE ############

fotosettore, 2022-04-15:
Hi folks !
For my needs i made a sketch to communicate between US2000C and ESP32.
I saw that new pylontech batteries does not needs the change of 1200/115200 and them answer immediately after a little wakeup.
The sketch is very poor and minimal. This is only a valid start for best development.
The answers of consolle are shown on serial monitor of IDE and hardware serial port number 2 is used, that is totally free from any use.
Sorry for sketch sintax but i'm a visually impaired, so I often need to use capital letters so as not to get confused.
The schematic I put is a modification of that of Wibet10.
Enjoy !!
p.s. if sketch does not work, you must modify the pins of serial2.
You can made it in HatdwareSerial.cpp
[image]

creativtransfer, 2022-04-24:
@fotosettore I seem to get some consistent data after sending GetPowBuff, but how do I parse the data?

fotosettore, 2022-04-24:
You can simply follow the sketch of Wibet10, that contains all decode calls. My test was only an hardware test to communicate using a ESP32 without web server.

creativtransfer, 2022-04-24:
Thank you for your reply. Actually, I am doing this on a raspberry pi with an USB to RS232 Adapter. So I have to write my own parser. How does a valid reply look like? Can you please post, what kind of response I should get after sending the  GetPowBuff Code. 
As stated, I am getting some data, however I am not sure if this is something valid or if I will also need a Logic-Level Converter.
Thank you very much!

fotosettore, 2022-04-24:
these are some calls i made using SERIAL PORT MONITOR software
[https://www.eltima.com/products/serial-port-monitor/?gclid=Cj0KCQjw6pOTBhCTARIsAHF23fIkegr2oNlbpRLsIYYpPPFzVjnh94Te6wTvnVYyU2LGrxt4zNspqgEaAu-5EALw_wcB](url) 

[23/04/2022 16:10:16] Written data (COM6)	
    74 6f 74 61 6c 63 6f 75 6c 0a                     totalcoul.      	
[23/04/2022 16:10:16] Read data (COM6)	
    74 6f 74 61 6c 63 6f 75 6c 0a 0d 40 0d 0d 0a 55   totalcoul..@...U	
    6e 6b 6e 6f 77 6e 20 63 6f 6d 6d 61 6e 64 20 27   nknown command '	
    74 6f 74 61 6c 63 6f 75 6c 27 20 2d 20 74 72 79   totalcoul' - try	
    20 27 68 65 6c 70 27 0d 0a 0d 24 24 0d 0a 0d 70    'help'...$$...p	
    79 6c 6f 6e 5f 64 65 62 75 67 3e                  ylon_debug>     	
[23/04/2022 16:10:59] Written data (COM6)	
    67 65 74 70 77 72 20 31 0a                        getpwr 1.       	
[23/04/2022 16:10:59] Read data (COM6)	
    67 65 74 70 77 72 20 31 0a 0d 40 0d 0d 0a 35 33   getpwr 1..@...53	
    32 37 36 23 20 20 20 20 20 20 30 23 20 20 32 35   276#      0#  25	
    30 30 30 23 20 34 39 34 30 30 23 20 49 64 6c 65   000# 49400# Idle	
    23 20 4e 6f 72 6d 61 6c 23 20 4e 6f 72 6d 61 6c   # Normal# Normal	
    23 20 4e 6f 72 6d 61 6c 23 20 0d 0d 0a 33 35 35   # Normal# ...355	
    31 23 20 20 32 31 30 30 30 23 20 4e 6f 72 6d 61   1#  21000# Norma	
    6c 23 20 4e 6f 72 6d 61 6c 23 20 0d 0d 0a 33 35   l# Normal# ...35	
    33 34 23 20 20 32 31 30 30 30 23 20 4e 6f 72 6d   34#  21000# Norm	
    61 6c 23 20 4e 6f 72 6d 61 6c 23 20 0d 0d 0a 33   al# Normal# ...3	
    35 35 34 23 20 20 32 31 30 30 30 23 20 4e 6f 72   554#  21000# Nor	
    6d 61 6c 23 20 4e 6f 72 6d 61 6c 23 20 0d 0d 0a   mal# Normal# ...	
    33 35 35 32 23 20 20 32 31 30 30 30 23 20 4e 6f   3552#  21000# No	
    72 6d 61 6c 23 20 4e 6f 72 6d 61 6c 23 20 0d 0d   rmal# Normal# ..	
    0a 33 35 35 32 23 20 20 32 31 30 30 30 23 20 4e   .3552#  21000# N	
    6f 72 6d 61 6c 23 20 4e 6f 72 6d 61 6c 23 20 0d   ormal# Normal# .	
    0d 0a 33 35 33 35 23 20 20 32 31 30 30 30 23 20   ..3535#  21000# 	
    4e 6f 72 6d 61 6c 23 20 4e 6f 72 6d 61 6c 23 20   Normal# Normal# 	
    0d 0d 0a 33 35 35 32 23 20 20 32 31 30 30 30 23   ...3552#  21000#	
    20 4e 6f 72 6d 61 6c 23 20 4e 6f 72 6d 61 6c 23    Normal# Normal#	
    20 0d 0d 0a 33 35 34 32 23 20 20 32 31 30 30 30    ...3542#  21000	
    23 20 4e 6f 72 6d 61 6c 23 20 4e 6f 72 6d 61 6c   # Normal# Normal	
    23 20 0d 0d 0a 33 35 35 37 23 20 20 32 31 30 30   # ...3557#  2100	
    30 23 20 4e 6f 72 6d 61 6c 23 20 4e 6f 72 6d 61   0# Normal# Norma	
    6c 23 20 0d 0d 0a 33 35 35 39 23 20 20 32 31 30   l# ...3559#  210	
    30 30 23 20 4e 6f 72 6d 61 6c 23 20 4e 6f 72 6d   00# Normal# Norm	
    61 6c 23 20 0d 0d 0a 33 35 35 39 23 20 20 32 31   al# ...3559#  21	
    30 30 30 23 20 4e 6f 72 6d 61 6c 23 20 4e 6f 72   000# Normal# Nor	
    6d 61 6c 23 20 0d 0d 0a 33 35 35 36 23 20 20 32   mal# ...3556#  2	
    31 30 30 30 23 20 4e 6f 72 6d 61 6c 23 20 4e 6f   1000# Normal# No	
    72 6d 61 6c 23 20 0d 0d 0a 33 35 35 38 23 20 20   rmal# ...3558#  	
    32 31 30 30 30 23 20 4e 6f 72 6d 61 6c 23 20 4e   21000# Normal# N	
    6f 72 6d 61 6c 23 20 0d 0d 0a 33 35 35 38 23 20   ormal# ...3558# 	
    20 32 31 30 30 30 23 20 4e 6f 72 6d 61 6c 23 20    21000# Normal# 	
    4e 6f 72 6d 61 6c 23 20 0d 0d 0a 33 35 35 37 23   Normal# ...3557#	
    20 20 32 31 30 30 30 23 20 4e 6f 72 6d 61 6c 23     21000# Normal#	
    20 4e 6f 72 6d 61 6c 23 20 0d 0d 0a 30 23 0d 0d    Normal# ...0#..	
    0a 31 32 36 23 0d 0a 0d 43 6f 6d 6d 61 6e 64 20   .126#...Command 	
    63 6f 6d 70 6c 65 74 65 64 20 73 75 63 63 65 73   completed succes	
    73 66 75 6c 6c 79 0d 0a 0d 24 24 0d 0a 0d 70 79   sfully...$$...py	
    6c 6f 6e 5f 64 65 62 75 67 3e                     lon_debug>      	
[23/04/2022 16:11:15] Written data (COM6)	
    70 77 72 73 79 73 0a                              pwrsys.         	
[23/04/2022 16:11:15] Read data (COM6)	
    70 77 72 73 79 73 0a 0d 40 0d 0d 0a 20 50 6f 77   pwrsys..@... Pow	
    65 72 20 53 79 73 74 65 6d 20 49 6e 66 6f 72 6d   er System Inform	
    61 74 69 6f 6e 0d 0d 0a 20 2d 2d 2d 2d 2d 2d 2d   ation... -------	
    2d 2d 2d 2d 2d 2d 2d 2d 2d 2d 2d 2d 2d 2d 2d 2d   ----------------	
    2d 2d 2d 2d 2d 2d 2d 2d 2d 2d 0d 0d 0a 20 53 79   ----------... Sy	
    73 74 65 6d 20 69 73 20 69 64 6c 65 0d 0d 0a 20   stem is idle... 	
    54 6f 74 61 6c 20 4e 75 6d 20 20 20 20 20 20 20   Total Num       	
    20 20 20 20 20 20 20 20 20 3a 20 33 20 20 20 20            : 3    	
    20 20 20 20 0d 0d 0a 20 50 72 65 73 65 6e 74 20       ... Present 	
    4e 75 6d 20 20 20 20 20 20 20 20 20 20 20 20 20   Num             	
    20 3a 20 33 20 20 20 20 20 20 20 20 0d 0d 0a 20    : 3        ... 	
    53 6c 65 65 70 20 4e 75 6d 20 20 20 20 20 20 20   Sleep Num       	
    20 20 20 20 20 20 20 20 20 3a 20 30 20 20 20 20            : 0    	
[... 21 lines cut ...]
    20 20 6d 56 0d 0d 0a 20 4c 6f 77 65 73 74 20 76     mV... Lowest v	
    6f 6c 74 61 67 65 20 20 20 20 20 20 20 20 20 20   oltage          	
    20 3a 20 33 35 33 32 20 20 20 20 20 6d 56 0d 0d    : 3532     mV..	
    0a 20 48 69 67 68 65 73 74 20 74 65 6d 70 65 72   . Highest temper	
    61 74 75 72 65 20 20 20 20 20 20 3a 20 32 31 30   ature      : 210	
    30 30 20 20 20 20 6d 43 0d 0d 0a 20 41 76 65 72   00    mC... Aver	
    61 67 65 20 74 65 6d 70 65 72 61 74 75 72 65 20   age temperature 	
    20 20 20 20 20 3a 20 32 31 30 30 30 20 20 20 20        : 21000    	
[... cut ...]
    67 20 63 75 72 72 65 6e 74 20 20 20 20 3a 20 37   g current    : 7	
    35 30 30 20 20 20 20 20 6d 41 0d 0d 0a 20 52 65   500     mA... Re	
    63 6f 6d 6d 65 6e 64 20 64 73 67 20 63 75 72 72   commend dsg curr	
    65 6e 74 20 20 20 20 3a 20 2d 37 35 30 30 30 20   ent    : -75000 	
    20 20 6d 41 0d 0a 0d 43 6f 6d 6d 61 6e 64 20 63     mA...Command c	
    6f 6d 70 6c 65 74 65 64 20 73 75 63 63 65 73 73   ompleted success	
    66 75 6c 6c 79 0d 0a 0d 24 24 0d 0a 0d 70 79 6c   fully...$$...pyl	
    6f 6e 5f 64 65 62 75 67 3e                        on_debug>

creativtransfer, 2022-04-25:
Thank you.
 I have a cheap USB to RS232 converter such as this one: https://de.aliexpress.com/item/488848564.html
with an RJ11 connector crimped to the end. I am getting data, but compared to your output it's a mess. No readable ASCII characters. Using 115200baud, I get a reply which repeats itself but it looks like this. When I have some time, I'll try to add the logic-level converter and try again.

From what I see, you are not appending a checksum (CRC) to the command, just finishing with 0x0a

This is the reply I get (115200baud):
'!#�0���巷���#)#!#�9!%%=#7����@��DD4Ŀ��

received dezimal values
33
168
64
145
148
68
4
68
52
172
229
127
229
229
235
85
35
41
35
33
17
35
191
57
33
37
37
61
35
55
191
177
133
168
64
145
148
68
4
68
52
196
...

fotosettore, 2022-04-25:
Unlike other protocols, such as that of BMS SEPLOS, that of pylontech fortunately is a simpler and more immediate protocol but above all it responds with ascii characters, as for example do the MPP SOLAR inverters.
Pylon protocol does not need CRC but only a LF (0x0A).
I repeat : I don't know how rasp works but remember that if you communicate with a RS232, it is already a 5V signal.
If you have not good answers, before of all test commands on a pc, using the software I advided in last post.

44matrix, 2022-10-10:
> Hi, I built one of these monitors a couple of years ago and it worked fine with my UC3000 batteries as described in the original design using the RJ11 port. End of last year I fited a new UC3000C battery as master and had to convert to the 8 pin 'Console' port. I rewired the cable and it started working but was erratic. It then refused to read at all and I abandonned it until this month. On investigation, the RS232 signals were barely 8-10volts pk/pk and the MQTT was trying to reconnect every cycle of the program. I also discovered from the web that the UC2000/3000C batteries do not need the 1200/115200 codes and rate change to wake them, they are ready to communicate at 115200 on startup. I have commented out 'void wakeUpConsole and void switchBaud' sections of the code and added this into the void loop: - void loop() { // millis delay check below runs 'mqttLoop' every 30 seconds Client.loop(); if (millis() - lastMillis >= 30*1000UL) // >= 30 sec period { lastMillis = millis(); // reset ready for next loop mqttLoop(); } ArduinoOTA.handle(); server.handleClient(); timer.run();_ code from this point is original -- The erratic connection problem was fixed by including a 3.3v to 5v level shifter (cost 99p) between the ESP and the MAX3232 as in the attached diagram. This had the effect of increasing the RS232 signal voltages sent to the battery to around 12/15volts pk/pk, much more' in spec' and a very solid connection. There have been some changes to the original project that accomodate the new battery types since I last 'checked in' (November 2021) and reading the above comments, the discoveries I made in adapting my original build may help resolve some of the problems experienced by others. [image] Hope this helps others connecting to UC2000/3000C batteries. Usual repeat of warnings - proceed at your own risk!

Was having issue and getting ?????   - with setup for Pylontech US3000C - but using Max3232 and level converter as per schematic shown resolved my issue.

dxoverdy, 2022-11-07:
Can someone explain to me why those mini RS232 boards only have the +/- drilled?  It drove me insane getting it soldered.  (Didn't realise I had the temp on the iron too high from a previous job - took the pad straight off :(

44matrix, 2022-11-07:
Don't use that -
When I used that it did not work!

Only worked with the Sparkfun chip. Not sure why.

NS

dxoverdy, 2022-11-11:
OK I've done the circuit by @Wibet10 however the first time I powered up the MAX3232 burnt out.  Yes it was a cheap one but exactly the one in the drawing.

Taking the rest of the circuit out of the equation, if I put just 5V and GND from the D1 mini to a new MAX3232, my next one got hot as hell within a second or two.

If I run one on the 3V it doesn't get hot, suggesting fundamentally it's a 3V chip.  Because the MAX3232 is digital to analogue, can I presume putting the logic shifter on the RS232 side won't work - i.e. amplify/reduce the RS232 side, not the TTL?

I'm a bit at a loss now.  They are all sold as 3-5V chips but i've got heat out of a basic cheap piece of junk and also some serious heat when running at 5V out of this: https://www.ebay.co.uk/itm/353203562315

Thanks for any help.

44matrix, 2022-11-11:
This thing is simple - if you get Heat, then you have some wiring issue. I used cheap max3232 chip and did not work until I used the level shifter and exact rs232 chip that was used by original design.

I will send you the drawing.

Regards,
NS

dxoverdy, 2022-11-12:
Any drawing or advice gratefully received 👍

I’m not sure how I could get my test case of a separate d1 mini and two jumper wires from 5v and Gnd to a stand-alone max3232 (which got hot) wrong by the way 😂. Reading online it could be latching?

44matrix, 2022-11-12:
Hi All,
This worked fine for me and I used original chip SparkFun MAX3232 (could not solder the cheapo little Ebay Max3232 chip properly - so I gave up as I have sausage fingers).
I am using US3000C batteries (two so far will be adding #3 within next few days) - although I can see information about my batteries - sometimes it is very delayed via SolisCloud App - and it does not show Battery Temperature. (Had lots of fun getting all this into HTML web page from JSON output.

My little site shows the current data and only updates the data only with text being static just like original design. I aim to add Battery Charge count to my display to keep track of number of Charge cycle my batteries have endured so far. But with only two - so far I have seen the count @ ~60 and these have been in use for about 80-100 days so far.
So I used FETCH() function and had to enable CORS in the D1 code. every time I post data.
Qdos to original designer - I will enhance as much as possible and help anyone else if I can.

Regards,NS

Wibet10, 2022-11-12:
Hi, 
You seem to be having some of the same problems I saw when trying to get my original build going. See diagram Feb 2022.
Here are some of the empirical facts I discovered from my various attempts;

Not all D1's are the same (in circuitry and quality) - some have very feeble 3.3V regulators which barely run the ESP and most have components between the USB port and the 5v rail pins. Therefore running any tests of intefaces using the USB to power the circuit may well fail even if the circuit is wired correctly. Be aware!

All D1's have 3.3v data and serial pins, pulling them up to 5v is almost certain to eventually kill or damage them. I have tried using LEDs and resistors to buffer several types of 5v interfaces to 3.3v data pins but this is unreliable on most versions of D1's (I found that it may work for a while but it usually becomes unreliable). 

The external 5v supply is best fed directly onto the 5v pins of the ESP, not via the USB and the 5v supply needs a local electrolytic capacitor to smooth and buffer the supply. (sometimes a parallel ceramic 0.1uF is also needed)
The MAX3232 has a voltage pump circuit which will introduce current spikes onto the 5v supply and can be a cause of latching. These spikes may also account for some random ESP resets I saw before including the 5v supply capacitors.

The logic level shifter is the most reliable way I have found to interface the 3.3v data inputs/outputs of the ESP to the MAX3232 serial driver (and many other 5v powered interface e.g. relays, Solid State mains switches etc.). 
On this bidirectional version of the level shifter, the Ground connection is common to both sides but the 3.3V and 5v sides are isolated. 
In my diagram, the electrolytic and ceramic capacitors are present on the 3.3V supply to help the ESP's on board regulator cope with the extra demand of the level shifter & serial interface (I do this now as a matter of course).

Remember that this version of the MAX3232 has two separate TTL to RS232 interfaces on the board, one on the top and another on the underside. The +v and -v pads are thro-hole plated so that both sides are powered but the in and out connections are totally separate for the top and bottom circuits. Be careful to get the send/receive connected correctly (ESP into RS232 end will likely destroy one or both!).

The MAX3232 chip is a TTL(0v~5v) to RS232 converter that has a voltage pump incorporated to convert/boost the TTL input (0v~5v) to RS232 levels +v~0v~-v . The chips voltage spec quotes 3v to 5v which is why they are often sold as 3.3v compatible, the best ones probably are but the cheap ones -  it's just pot luck! 
I had some overheating problems with the first few MAX boards I tried before using the logic level shifters. The 4 off boards (8 circuits) I tried running at 3.3v, most just got hot or appeared dead. When the voltage pump stops switching it gets hot, this is what they refer to as latching eventually it fails - probably caused by insufficient voltage/power? 

When I did get one MAX 3232 working, powered via 3.3v from the ESP - the output voltages at the serial side was approximately +4v to -4v around the ground (0v) - 8v pk-pk - this is the very minimum for the RS232 spec. (it did not work reliably with my Pylon 3000 battery interface)
When connecting the MAX3232 to 5V and driving it via the level shifter from the ESP the RS232 voltages achived were +7.5v~0v~-7.5v - 15v pk-pk - this is comfortably in the RS232 spec. These voltages can be checked on an oscilloscope but they only change when the unit is sending or recieving data. (this configuration works with the Pylon 3000 & 3000C battery ok)

Lastly the 100ohm termination resistor - this terminates the serial line output of the RS232 and prevents any excessive voltage overshoot from the voltage pump in the MAX3232. Possibly another cause of latching and overheating? Serial transmission lines should always be terminated correctly.

After resolving the above issues I found that a few of the circuits on the MAX3232 boards I had been experimenting with did still work but it had been the factors above that made it appear that they had failed, some circuits were actually dead tho', I don't know if I had killed them while experimenting or they were DOA?
If you have correctly powered the MAX3232 with 5v, connected it up the right way around, drive it via a level shifter from the ESP, have terminated the serial line with 100ohms and have adequately buffered the 5v & 3.3v supplies then it should not get hot (when idle or when signalling) and should actually send balanced +v~0v~-v signals into the 100ohm resistor.

I have not tried any other type of MAX3232 driver board other than the basic type shown in my diagram but if they are using the same chip they will have the same limitations. 
My very cheap MAX3232 has been running reliably now for more than 12 months in its latest incarnation on the 3000C batteries (another 18 months before that on the 3000 series batteries) so it appears to be stable and reliable.

I hope that these pointers may help you locate your problem areas and get your interfaces working.

dxoverdy, 2022-11-12:
Thank you for your comprehensive guides and explanations.

One thing that probably explains why my first mini max3232 didn’t communicate, I’d burnt off the pad so just soldered to the underside thinking it was one of the same.  Lol.

In the first instance before all this I built according to the original spec as part of this project.  I used all 3.3v and an all in one rs/ttl board.  That same board is pictured later.  When using 3.3 logic it didn’t get hot but I only got ????.  I tried the 5v supply but 3.3v serial pins and that caused it to get mad hot and equally didn’t work which led me to this thread.

I built your circuit on some prototyping board including the caps and what not but the small max3232s just got mad hot, I totally cooked one of them.  When I took everything out of the equation and just gave it 5v and gnd from the d1 it got hot, indicating like you say, that the 5v out probably can’t be trusted.  And I am using budget spec d1 minis.  £3 ish.

I’ve just prototyped the following, I’ve not tested it yet, but essentially that same ttl daughterboard is now on the same 5v as the d1 but with the logic level introduced on tx/rx.  Obviously this is notwithstanding your supply advice and I’m not sure if this board terminates the serial as advised… but it’s flashing tx continuously (probably part of the sketch but I’ve not started looking at rewriting that yet until I at least had a sign of life) but it is running cold and my multimeter sees a blip on the rx pin of the rj11.

Which incidentally is probably the cause of all my grief in the first place as the original pic on here indicated straight through pins 2 and 3 however that’s been confirmed as being wrong and they should have crossed.

[image]

I’ll drive this over to my dads later and give it a quick test.  I’m on an AlphaESS system so I leverage modbus/rs485 for all the info, he wants to do the same on this.

Thanks again

hardweb-it, 2023-09-30:
I've two US2000C.
Using a Wemos D1 mini clone, I was getting al zeroes at /jsonOut values.
I only switch the 3v3 to 5v to power MAX2323, without any logic converter addon, and then it worked!

{"soc": 73, "temp": 32000, "currentDC": 22489, "avgVoltage": 50587, "baseState": "Charge", "batteryCount": 2, "powerDC": 1137, "estPowerAC": 1205, "isNormal": true}