irekzielinski/Pylontech-Battery-Monitoring #5: US2000C and US3000C models.

JosekeyRandom, 2021-04-16:
Good morning, could you add communication with the new US2000C and US3000C models? Apparently an rj45 port is already used for console mode but something must have changed in the protocol because the only software that works is MultiSIBControl and it had to make a small update to recognize them http://www.multisibcontrol.net/

Would it be possible to add support for the new models?

Here is information about the connections https://powerforum.co.za/topic/7901-pylontech-us3000c/

Thanks.

Good morning, could you add communication with the new US2000C and US3000C models? Apparently an rj45 port is already used for console mode but something must have changed in the protocol because the only software that works is MultiSIBControl and it had to make a small update to recognize them http://www.multisibcontrol.net/

Would it be possible to add support for the new models?

Here is information about the connections https://powerforum.co.za/topic/7901-pylontech-us3000c/

Thanks.

PD: I still do not have any of these new models but they are probably the ones that will be sold when the stock of old models runs out.

mihaigsm2003, 2021-04-16:
Hi there!

I gest have ready partially software for esp8266 and I use to my server :

https://gsx.ro/d/RijerOxMz/casa-haotik?orgId=1&refresh=1m

irekzielinski, 2021-08-05:
Hi guys. I don't have a C model so can't experiment.
@mihaigsm2003 - which battery do you have? Is it a "C" model?
Can you share your wiring diagram and if you had to make any other changes (for example is the protocol the same)?

Thanks!
Irek

Tinbum1, 2021-08-10:
[image]

[image]

 Both are RJ45 sockets, not sure about the protocol.

dmqgit, 2021-12-16:
Everything works with the newer C-Series (hopefully still of interest). Protocol did not change. Just change to the socket to RJ45. Thanks @irekzielinski for the code.

mihaigsm2003, 2021-12-16:
i have pinout and work very good, but i need to read data from battery and place via mqtt in grafana. need to read each cell automatically at every 30 seconds

irekzielinski, 2021-12-16:
> Everything works with the newer C-Series (hopefully still of interest). Protocol did not change. Just change to the socket to RJ45. Thanks @irekzielinski for the code.

Thanks for the info - would you mind providing some deails of the wiring you used?
I will be happy to update the documentation.

dmqgit, 2021-12-16:
Sure.

RJ45 cable has 8 wires (outer right is full brown color based on EIA/TIA-568-B (version EIA/TIA-568-A is different)):

3 (white-green) = rx
6 (green) = tx
8 (brown) = gnd

irekzielinski, 2021-12-16:
> Sure.
> 
> a standard Ethernet - RJ45 cable has 8 wires (outer right is full brown color based on EIA/TIA-568-B (version EIA/TIA-568-A is different)):
> 
> 3 (white-green) = rx 6 (green) = tx 8 (brown) = gnd

Thanks. Just to be 100% sure:
> 3 (white-green) = rx 
Is this RX of the battery (so should be connected to TX of Wemos), or the other way around?

dmqgit, 2021-12-16:
No, the other way around :)

3 (white-green) tx.pylontechtech = rx.wemos

irekzielinski, 2021-12-16:
> No, the other way around :)
> 
> 3 (white-green) tx.pylontechtech = rx.wemos

Thanks, updated the front page with this information. And closing this issue as support it now confirmed :)

closed 2021-12-16 by irekzielinski

mduchain, 2022-01-02:
Just a feedback; I have US2000B Plus manufactured September 2021 and it doesn't need the wakeup sequence to actually get the "pylon>" prompt. No need to drop to 1200 baud or send that byte sequence to communicate.
Newer battery firmware probably. Here are the FW versions reported by the battery "info" command:

Device address      : 1
Manufacturer        : Pylon
Device name         : US2KBPL
Board version       : PHANTOMSAV10R03
Main Soft version   : B67.10.0.1
Soft  version       : V3.0
Boot  version       : V2.2
Comm version        : V2.0
Release Date        : 21-06-07

Specification       : 48V/50AH
Cell Number         : 15
Max Dischg Curr     : -100000mA
Max Charge Curr     : 102000mA
EPONPort rate       : 1200
Console Port rate   : 115200

irekzielinski, 2022-01-02:
> Just a feedback; I have US2000B Plus manufactured September 2021 and it doesn't need the wakeup sequence to actually get the "pylon>" prompt. No need to drop to 1200 baud or send that byte sequence to communicate. Newer battery firmware probably. 

Interesting, is the "wake up" sequence doing any problems on your system? Or does it work with it?

mduchain, 2022-01-02:
No, the wake up doesn’t do any harm. The code was working as intended with the wakeup sequence.

I noticed that when I reset the battery without restart of the ESP, I would get the pylon prompt immediately without going through the wakeup sequence.

I then removed the wakeup code altogether and everything kept working fine, even after multiple reboot/power-toggles of both ESP and battery.

The batteries’ console serial speed is not configurable and fixed at 115200 according to the manuals I received together with the battery.

 

I do get some noise (bad chars) when not sending commands for a while, but I believe this is due to the MAX3232 working at 3.3V (gathered from a few posts around the subject).

I avoid this by sending a few CR/LF before sending the actual command – an alternative wakeup sequence of sorts.