irekzielinski/Pylontech-Battery-Monitoring #9: only see "????" on "power" , "help" ... timezone setting?

a68k, 2022-04-20:
not sure what is wrong ...
I can see the page with the uptime (where can this be set to show correct time?)
but when opening "help" or "power" I wait some time and only see a page with "????"

a68k, 2022-04-22:
thanks for competent help

not

gerdluthe, 2022-05-19:
Check your wemos, i'd used an wemos mini pro and the communication with my pylontech ended as yours in an ????
so i'd used an wemos mini d1, maybe a clone one and with this one i can comminicate with my pylontechs

RomanSar779, 2023-06-11:
Hallo @a68k, konntest du dein Problem mit einen anderen ESP lösen? Hab auch nur ????

luftdieb, 2024-12-15:
Same problem on my side. Cheap Wemos D1 Mini won't work. Webinterface show startpage but wasn't able to show battery status. After using Wemos from Delivery (Amazon), it works directly. It costs me 4h troubleshooting.