irekzielinski/Pylontech-Battery-Monitoring #10: RS485 ?

a68k, 2022-04-21:
Can it also work with RS485?

I have a ROSEN SOLAR 10kWh powerwall and some say, has the same BMS inside.
But has no serial port...

a68k, 2022-05-11:
20 days later ... no reply.

speculatrix, 2022-05-12:
> 20 days later ... no reply.

I'd get a refund on the hugely expensive support contract you paid for?

a68k, 2022-05-12:
who are you?
what is that for an answer???
showing yourself how dumb you are?

did it help/solve in the issue?
[x] no

did it show competence?
[x] no

speculatrix, 2022-05-12:
> who are you? what is that for an answer??? showing yourself how dumb you are?
> 
> did it help/solve in the issue? [x] no
> 
> did it show competence? [x] no

Entitled much? Arrogant much? Rude much?

a68k, 2022-05-12:
ask yourself in the mirror

speculatrix, 2022-05-12:
> ask yourself in the mirror

You've made an astonishing contribution to humanity here on github
https://github.com/a68k?tab=repositories

Oh wait, you've done nothing, zero, zilch.

a68k, 2022-05-12:
oh you are such a smart guy muhahaa
you are drunk? on drug? or from nature an a.h?
Go on, show the world more of you

gerdluthe, 2022-05-19:
check google, there are many tutorials to communicate via 485 with the pylontechs, this program I think won't run with RS485. your comments above are not useful !!!