irekzielinski/Pylontech-Battery-Monitoring #12: SparkFun MAX3232

ch-pk, 2022-09-23:
Hi.
Wanted to ask if the circuit only works with the SparkFun MAX3232 or also with other RS232-TTL converters.

gerdluthe, 2022-09-23:
Give it a try, there's an max3232 onboard the aliexpress one's

ch-pk, 2022-09-23:
I tried this one from Ali but get no data.

gerdluthe, 2022-09-23:
i'd used an blank max3232 without any problems, you'd checked all the wiring stuff? wifi connection is good?

ch-pk, 2022-09-23:
WIFI is ok because I can access the site. But there is no data coming from Pylontech. I will check everything again otherwise solder a new one.
It is US2000C version.

gerdluthe, 2022-09-23:
mine is the us2000c too. good luck

ch-pk, 2022-09-23:
Thanks. Will try again.

fourofspades, 2023-06-01:
I'm having trouble sourcing that boards. Everywhere out of stock and many months leadtime.

This board appears to have USB, so presumably some form of chip onboard?

Could I use this without the need for max232?

FREENOVE ESP8266 Development Board (Compatible with Arduino IDE), ESP-12S Onboard Wi-Fi, MicroPython C Code, Detailed Tutorial, Example Projects https://amzn.eu/d/6RC3Umo

commander-bln, 2023-06-05:
hi

first- you need a Max3232 for converting the serial2ttl for the console port of the pylontech
sec. you can use any esp8266 or better the esp32 with 4MB 
if using only the chip you need a converter for programming, if it is a deleopment Board you have the converter on board and therfor is the USB plug.

I used a "DOIT ESP32 DevKit V1" from Amazon  with an Ardafriut MAX3232 an works fine on pylontech US5000 (2X)
if needed i can upload an Fritzing layout??

bartixmayo, 2023-10-17:
In other project I had to try 4 (four!) noname/cheap serial2ttl boards until I have got working one.

Rustimation, 2024-11-14:
Same thing here: 
Treedix Mini RS232 auf TTL MAX3232 (Amazon) costs about a quarter of the SparkFun product. It works in general. However you have to unplug the board (at least the RX line) from the D1 Mini Microcontroller in order to be able to upload a sketch. The Sparkfun thing didn't have this problem.