irekzielinski/Pylontech-Battery-Monitoring #13: Cycle Count

albundy0815, 2022-09-26:
I guess it would be very helpful to get the "cylce counts" also availble via MQTT. It's in the command "stats" but it seems, that this value is not inside the MQTT function. 
It would be great to get this function... Thanks

tomba0815, 2022-12-05:
Hello Ireneusz,
I also wanted to ask if you could also query the number of cycles.

tomba0815, 2022-12-05:
I thinks that's not so easy, at first you have to pull the correct values ....

tomba0815, 2022-12-06:
I'm currently using the not-so-elegant way to do this via a request. I query the website via iobroker, then I use RegEx to extract the number of cycles.
This works, but is not as convenient as with MQTT.

albundy0815, 2022-12-06:
I guess the easiest thing for all, would be to implement it directly, that it can be provided by MQTT. But... still no response from the developer and my scripting experience is more than little. ;(

commander-bln, 2023-02-06:
i have tried it with my new pylontech US5000, Sketch works fine, i have updated the scetch to use with ESP32 and the first change to get SOC in MQTT, it works

for SOC i can help...if wanted

tomasvida, 2023-02-22:
> i have tried it with my new pylontech US5000, Sketch works fine, i have updated the scetch to use with ESP32 and the first change to get SOC in MQTT, it works
> 
> for SOC i can help...if wanted

Could I ask for a sample of the modified code? I have high voltage Pylontech battery, I need to modify the code, so I would appreciate some inspiration.

commander-bln, 2023-02-22:
hello tomasvida
can i send you my *.ino file diretly? or shoud i post it here??

albundy0815, 2023-02-22:
> hello tomasvida can i send you my *.ino file diretly? or shoud i post it here??

if you paste it here, the whole community would be able to access.

tomasvida, 2023-02-22:
> hello tomasvida can i send you my *.ino file diretly? or shoud i post it here??

If it is possible, please be said to be here so that the others in the community can be black experiences. It will help us very much, and thank you for your work and willingness!

commander-bln, 2023-02-22:
here is my *ino file works on esp32 on pylontechUS5000 on RS232 Port

`#include <WiFi.h>
#include <ESPmDNS.h>
#include <ArduinoOTA.h>
#include <WebServer.h>
#include <WebServer.h>
#include <SimpleTimer.h>
#include <TimeLib.h> //https://github.com/PaulStoffregen/Time
#include <ntp_time.h>
#include <circular_log.h>

//IMPORTANT: Specify your WIFI settings:

#define WIFI_SSID "YourSSID"
#define WIFI_PASS "YourPass"

//IMPORTANT: Uncomment this line if you want to enable MQTT (and fill correct MQTT_ values below):
#define ENABLE_MQTT

#ifdef ENABLE_MQTT
//NOTE 1: if you want to change what is pushed via MQTT - edit function: pushBatteryDataToMqtt.
//NOTE 2: MQTT_TOPIC_ROOT is where battery will push MQTT topics. For example "soc" will be pushed to: "home/grid_battery/soc"
#define MQTT_SERVER        "192.168.100.5"
#define MQTT_PORT          1883
#define MQTT_USER          "YourUser"
#define MQTT_PASSWORD      "YourUserPassword"
#define MQTT_TOPIC_ROOT    "Pylontech/US5000/"  //this is where mqtt data will be pushed
#define MQTT_PUSH_FREQ_SEC 2  //maximum mqtt update frequency in seconds

#include <PubSubClient.h>
WiFiClient espClient;
PubSubClient mqttClient(espClient);
#endif //ENABLE_MQTT

char g_szRecvBuff[7000];

WebServer server(5042); // Port for Webserver, for Standard take 80
SimpleTimer timer;
circular_log<7000> g_log;
bool ntpTimeReceived = false;
int g_baudRate = 0;

void Log(const char* msg)
{
  g_log.Log(msg);
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT); 
  digitalWrite(LED_BUILTIN, HIGH);//high is off
  
  // put your setup code here, to run once:
  WiFi.mode(WIFI_STA);
  WiFi.persistent(false); //our credentialss are hardcoded, so we don't need ESP saving those each boot (will save on flash wear)
  WiFi.hostname("ESP32toPylontechBattery");
  WiFi.begin(WIFI_SSID, WIFI_PASS);

  for(int ix=0; ix<10; ix++)
  {
    if(WiFi.status() == WL_CONNECTED)
    {
      break;
    }

    delay(1000);
  }

  ArduinoOTA.setHostname("YourHostname");
  ArduinoOTA.begin();
[... 678 lines cut ...]
  //next: read data from battery and send via MQTT (but only once per MQTT_PUSH_FREQ_SEC seconds)
  static unsigned long g_lastDataSent = 0;
  if(mqttClient.connected() && 
     os_getCurrentTimeSec() - g_lastDataSent > MQTT_PUSH_FREQ_SEC &&
     sendCommandAndReadSerialResponse("pwr") == true)
  {
    static batteryStack lastSentData; //this is the last state we sent to MQTT, used to prevent sending the same data over and over again
    static unsigned int callCnt = 0;
    
    parsePwrResponse(g_szRecvBuff);

    bool forceUpdate = (callCnt % 20 == 0); //push all the data every 20th call
    pushBatteryDataToMqtt(lastSentData, forceUpdate);
    
    callCnt++;
    g_lastDataSent = os_getCurrentTimeSec();
    memcpy(&lastSentData, &g_stack, sizeof(batteryStack));
  }
  
  mqttClient.loop();
}

#endif //ENABLE_MQTT`

i will try to get the "stat" Data also, but this will not be so easy as change a good code

commander-bln, 2023-02-22:
[PylontechMonitoring-ESP32.txt](https://github.com/irekzielinski/Pylontech-Battery-Monitoring/files/10807248/PylontechMonitoring-ESP32.txt)

i hope this is better

rickyrickuk, 2023-10-26:
> [PylontechMonitoring-ESP32.txt](https://github.com/irekzielinski/Pylontech-Battery-Monitoring/files/10807248/PylontechMonitoring-ESP32.txt)
> 
> i hope this is better
> 
> i will try to get the "stat" Data also, but this will not be so easy as change a good code

Did you have any luck with the stat data? Also interested to get some of the info from this command such as Cycle times, LT Cnt, etc.
Thanks

commander-bln, 2023-10-27:
@rickyrickuk  
sorry, not yet 

i hope now in the darker time i have more time to try a little bit
i will post here when i find a good solution

rickyrickuk, 2023-10-27:
> @rickyrickuk sorry, not yet
> 
> i hope now in the darker time i have more time to try a little bit i will post here when i find a good solution

Thanks for the reply.
I am having a look myself too to see if I can put something together :-)

One thing I noticed, is there is a parameter called _Status Cnt_ in the stat command results, wondered if this is the number of times the stat command has been run and if so, this would go very high if we keep sending the command to the battery.
Not sure if this will trigger anything in the event of a warranty claim?

commander-bln, 2023-10-27:
@rickyrickuk 

stat shows 1st Batt
if have more then 1 then 
stat 1 is the first
CYCLE Times     :      163
stat 2 is the second
CYCLE Times     :       92
stat 3 is the third
CYCLE Times     :       55
...my data over the net....

Status Cnt.     :      629  <- this is my first battery (started on 1/23)  

one example snip for
stat 1

@
Device address           1
Data Items      :     1281
HisData Items   :     1793
Charge Cnt.     :        0
Discharge Cnt.  :        0
Charge Times    :    51098
Status Cnt.     :      629
Idle Times      :    61779
COC Times       :        0
COC2 Times      :        0
DOC Times       :        0
DOC2 Times      :        0
COCA Times      :        0
DOCA Times      :        0
SC Times        :        2
Bat OV Times    :        0
Bat HV Times    :      360
Bat LV Times    :        0
Bat UV Times    :        0
Bat SLP Times   :        0
Pwr OV Times    :       17
Pwr HV Times    :      779
Pwr LV Times    :        0
Pwr UV Times    :        0
Pwr SLP Times   :        0
COT Times       :        0
CUT Times       :        0
DOT Times       :        0
DUT Times       :        0
CHT Times       :        0
CLT Times       :        0
DHT Times       :        0
DLT Times       :        0
Shut Times      :        1
Reset Times     :       16
RV Times        :        0
Input OV Times  :        0
SOH Times       :        0
BMICERR Times   :        0
CYCLE Times     :      163
SOH             :       99
Pwr Percent     :       21
Pwr Coulomb     : 75424820
Dsg Cap         : 16364081
HT@0.5C Cnt     :        0
LT@0.5C Cnt     :        0
HT Cnt          :        0
LT Cnt          :        0
LV Cnt          :       82
LifeWarn Times  :        0
LifeAlarm Times :        0
Command completed successfully
$$
pylon>

rickyrickuk, 2023-10-27:
@commander-bln
Thanks for the outputs.
I tried a few stat messages over a period of an hour or so and the StatusCnt value seemed to reset to 0, maybe its a count per 1/2 hr or something?