irekzielinski/Pylontech-Battery-Monitoring #14: read cells voltage

mihaigsm2003, 2022-10-22:
I need to read voltage from each cell and send via mqtt, how to add to code?

commander-bln, 2023-02-06:
i also need the cells voltage, the scetch only use the "pwr" Data (works fine), what have to be change to also get all the data from "bat" ??

have anybody tried this ???

gitisgreat2023, 2024-11-12:
I implemented reading out the individual cell voltages and saving them to MQTT topics using Node Red.

Here's the code
`[
    {
        "id": "210ecc232aa515d1",
        "type": "tab",
        "label": "Pylontech",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "1a2b3c4d5e6f7g8h",
        "type": "http request",
        "z": "210ecc232aa515d1",
        "name": "Get Battery Data",
        "method": "GET",
        "ret": "txt",
        "paytoqs": "ignore",
        "url": "http://pylon.fritz.box/req?code=bat",
        "tls": "",
        "persist": false,
        "proxy": "",
        "insecureHTTPParser": false,
        "authType": "",
        "senderr": false,
        "headers": [],
        "x": 890,
        "y": 540,
        "wires": [
            [
                "2b3c4d5e6f7g8h9i"
            ]
        ]
    },
    {
        "id": "2b3c4d5e6f7g8h9i",
        "type": "function",
        "z": "210ecc232aa515d1",
        "name": "Process Battery Data",
        "func": "// Get the HTML input from the HTTP request\nlet htmlInput = msg.payload;\n\n// Use a regular expression to extract voltage values from the provided HTML\nlet voltageRegex = /\\d+\\s+(\\d+)\\s+-?\\d+\\s+\\d+\\s+\\w+\\s+\\w+\\s+\\w+\\s+\\w+\\s+\\d+%\\s+\\d+\\s+\\w+/g;\nlet voltages = [];\nlet labels = [];\nlet match;\n\n// Loop through all matches and extract the voltage (second column)\nlet index = 0;\nwhile ((match = voltageRegex.exec(htmlInput)) !== null) {\n    if (match[1] !== '-') { // Check if the voltage is not absent\n        voltages.push(parseInt(match[1]));\n        node.warn(\"C\" + (index + 1) +' '+voltages[index]+'mV')\n        labels.push('C' + (index + 1) + ' ' + voltages[index] + 'mV'); // Create labels for each cell\n        //labels.push(`C${index + 1}${voltages[index]}`); // Create labels for each cell\n        index++;\n    }\n}\n\n\n\n// Calculate the minimum and maximum voltages\nlet minVoltage = Math.min(...voltages);\nlet maxVoltage = Math.max(...voltages);\n\n// Calculate the average voltage\nlet averageVoltage = voltages.reduce((sum, voltage) => sum + voltage, 0) / voltages.length;\n\n//node.warn(\"voltages: \" + voltages)\n\n// Prepare the title for the bar chart with dynamic values divided by 1000\nmsg.title = `Avg${(averageVoltage / 1000).toFixed(2)}V,Min:${(minVoltage / 1000).toFixed(2)}V,Max:${(maxVoltage / 1000).toFixed(2)}V,delta:${((maxVoltage - minVoltage)).toFixed(0)}mV`;\n\nmsg.payload = [{\n    \"labels\": labels,\n    \"data\": [voltages],\n    \"series\": [msg.title] \n}];\n\n//node.warn(\"title: \" + msg.title)\n\nconst messages = {};\n// Loop through each line and extract the voltage\nfor (let i = 0; i < 15; i++) {\n    messages[`cellv_${i + 1}`] = voltages[i];\n}\nmessages['cellv_min'] = minVoltage;\nmessages['cellv_max'] = maxVoltage;\nmessages['cellv_delta'] = maxVoltage - minVoltage;\nmsg.mqtt = [messages];\n\nmsg.mqtt = [...voltages,minVoltage,maxVoltage,maxVoltage - minVoltage];\n\n// Set dynamic ymin and ymax for auto-zooming\nmsg.ui_control = {\n    ymin: minVoltage - 2, // Optional padding\n    ymax: maxVoltage + 2,  // Optional padding\n    title: msg.title\n};\n\n// Return the modified message\nreturn msg;",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 1140,
        "y": 540,
        "wires": [
            [
                "3c4d5e6f7g8h9i0j",
                "967347af5d917976",
                "86c188e82de695c4"
            ]
        ]
    },
    {
        "id": "3c4d5e6f7g8h9i0j",
        "type": "ui_chart",
        "z": "210ecc232aa515d1",
        "name": "Voltage Bar Chart",
        "group": "55d02026703d8522",
        "order": 0,
        "width": 0,
        "height": 0,
        "label": "Battery Voltages",
        "chartType": "bar",
        "legend": true,
[... 364 lines cut ...]
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    },
    {
        "id": "e5b1aeb5977e428a",
        "type": "ui_tab",
        "name": "Home",
        "icon": "dashboard",
        "order": 1,
        "disabled": false,
        "hidden": false
    }
]`