irekzielinski/Pylontech-Battery-Monitoring #17: New firmware 2.8 us2000c, not works.

GuillermoElectrico, 2023-01-07:
With the new version of firmware 2.8 of the us2000c batteries no longer works. It seems that the port is now initialized to 115200 by default, but the problem I see is that it doesn't seem to accept the newline every time a request is sent.

If I use esp and send a request, for example, help, it does nothing, but if I disconnect esp and connect directly to a computer and do the same query using terminal1.9b (with +CR/LF enabled) I get an error of Unrecognized command and it appears to me that the query consists of two help followed by no spaces, that is, "helphelp".

I've gone through the code and tried alternatives to \n or even adding \r to no avail.

It has been working perfectly until I updated to version 2.8 (I added a new battery and upgraded the existing ones).

I have tried everything, including disabling the wakeUpConsole so that it is at a fixed 115200, but without results.

Any idea what may be going on?

gerdluthe, 2023-01-07:
What about downgrading all batterys to the working one you used before.
what's the benefits of having an actual 2.8 firmware as pylontech never released an info whats new in the firmware releases

GuillermoElectrico, 2023-01-07:
I don't know about the improvements, but what I do know is that a us2000b has bloated me.

I have 4 us2000c and 6 (well now 5) us2000b plus. When assembling a new us2000c, which came with the latest version and as recommended, it must be set as master, one of the us2000b has failed, it seems overloaded or similar. To avoid further problems, they recommended me to update the rest of us2000c to 2.8 (us2000b were already with 3.4).

Since the update, when loading, it doesn't do weird things anymore and it's stable. But now I have found the problem that the esp8266 does not work in any us2000c (in the us2000b it does, but since the set exceeds 8, I cannot use it as a master).

But it's weird, from a PC it works without a problem (at 115200) but the esp8266 doesn't, and as I mentioned, it seems like a line break termination problem or similar. But I don't give up on it.

GuillermoElectrico, 2023-01-14:
Found the problem, it is not a problem of the code, but of the hardware. Now with the new version of the firm, the MAX3232 converter that has been working perfectly at 3.3V powered from the NodeMCU, no longer works. Generates weird and erroneous characters.

It has been to put a level converter and feed it at 5V, and it works fine from the first moment.

The funny thing is that with the previous versions of the firm, and US2000B batteries, it had no problems, but with the new US2000C it is more sensitive.

https://github.com/irekzielinski/Pylontech-Battery-Monitoring/issues/5#issuecomment-1003703097

As I have searched, there have already been others with the same problem. My solution, to feed the MAX3232 to 5V and mount a level converter between the ESP8266.

closed 2023-01-14 by GuillermoElectrico