irekzielinski/Pylontech-Battery-Monitoring #22: Software und/or hardware modifications for Pylontech US3000C

alexandergeiler, 2023-07-28:
Hi,

I use [irekzielinski](https://github.com/irekzielinski) [Pylontech-Battery-Monitoring](https://github.com/irekzielinski/Pylontech-Battery-Monitoring) solutions since years at 2 different locations.
At the first location, I have a Pylontech US2000 at the second location I have a Pylontech US3000.
The solution is working fine.
I did some small modifications, SYSLOG support, change the OTA update procedure, diffrent landing page, ... but the hole functionallity is still from [irekzielinski](https://github.com/irekzielinski)

I ordered a new Pylontech US3000C to replace the Pylontech US2000.
I tried to connect it to my existing WEMOS but a communications was not possible.

I didn't see that irekzielinski updated his documentation and found the the description of how to wire it up now at:
[issue5](https://github.com/irekzielinski/Pylontech-Battery-Monitoring/issues/5)

Also after changing the wiring, I am not able to communicate with the Pylontech US3000C.
[image]

According to:
[issue1](https://github.com/irekzielinski/Pylontech-Battery-Monitoring/issues/1)
[Wibet10](https://github.com/Wibet10) added a
IIC I2C Logic level converter bi-directional module
and changed the source code

According to:
[issue5](https://github.com/irekzielinski/Pylontech-Battery-Monitoring/issues/5)
The old code is still working without a change.
[mduchain](https://github.com/mduchain) identified that the "drop to 1200 baud or send that byte sequence to communicate" is not needed anymore.
And few days later [mduchain](https://github.com/mduchain) write "I avoid this by sending a few CR/LF before sending the actual command – an alternative wakeup sequence of sorts."

---
Now my question?
Is anyone using the orginal source code from [Pylontech-Battery-Monitoring](https://github.com/irekzielinski/Pylontech-Battery-Monitoring) to communicate with a Pylontech US3000C?
Is it working without a IIC I2C Logic level converter bi-directional module?

commander-bln, 2023-07-28:
I use the original code with an ESP32 Dev Kit V1 without level converter, only TRANSCEIVER BREAKOUT - MAX3232 from SparkFUN and this works fine with 3X US5000 on Port Console

alexandergeiler, 2023-07-28:
@commander-bln your answer is really helpful.
The "wakeUpConsole();" is still active in the code?

I have a cheap MAX3232 from Aliexpress that was working with my Pylontech 2000. May be I should try the MAX3232 from SparkFUN.

commander-bln, 2023-07-28:
hello
your name sounds like german??

I have 3 "TRANSCEIVER BREAKOUT - MAX3232 from SparkFUN" and all worked on breadboard.
i heard sometimes the user had problems with the transceiver and tryed multiple , till they get lucky.

[image]

alexandergeiler, 2023-07-28:
Hi [commander-bln](https://github.com/commander-bln),
ja, Deutscher :-)
Ich habe einmal den "orginal " MAX3232 from SparkFUN an meiner US3000 und einen "noname" MAX3232 an meiner US2000 (die ich gerade versuche zu tauschen).
Ich teste das mal mit dem orginal MAX3232 from SparkFUN
Danke für Dein Bild!

commander-bln, 2023-07-28:
OK, ich hatte auch am Anfang Fehler, deshalb hatte ich mit Fritzing PCB und Schaltplan gezeichnet, und dann hatte ich auch schnell die Fehlverdrahtung auf dem Breadboard entdeckt. 
[image]
[image]

Auf dem Board sind auch 4-5 Kondensatoren bestückt, vielleicht macht das den Transceiver Stabiler..

alexandergeiler, 2023-07-29:
Ich schaue nochmal. Auf die schnelle habe ich es nicht hinbekommen.
Danke erstmal.
Eine US5000 ist baugleich mit einer US300C?

guybw, 2023-07-29:
If it helps, someone has forked this to support the us5000 and us3000c so that might help ?

alexandergeiler, 2023-07-29:
I tested the fork from Wibet10 too.
My US2000 and my US3000 are connected via the "console" port.
Several people seems to be using the B/RS485 port.

Can someone that is using a US3000C decibe, if he is using the "console" or the "B/RS485" port.

Startertrek, 2023-07-30:
I am using console port. Connected are 1 x US2000c and 1 x US3000c, both are reading stable

commander-bln, 2023-07-30:
I am using 3x us5000 on Port  console with no Problems...using esp32.with converter tll2seitial Von Unterwegs gesendet.

alexandergeiler, 2023-07-31:
@Startertrek, nice to know that it works for you with a US3000C.

So you use the unchanged code from irekzielinski. Including the wakeup call with 1200 baud?

Are you using the SpaarkFun MAX3232 transceiver or a "none".
Are you using an I2C IIC logic level converter 5V~3.3V?

Last question, all your jumper at the US3000C are looking to the ADD desciption at the batterie?

[image]

commander-bln, 2023-08-05:
Here is my Breadboard Wiring that Works with US5000 and the standard code
ESP32 Dev Kit V1 + MAX3232 and konnector RJ45 , i use standard RJ45 cable

the only thing i have make with my First Battery, i connect with Battery View 3.0.24 to look for Firmware/Update

[image]

alexandergeiler, 2023-08-25:
Hi all,

it was a layer 8 problem.
It is now working with a SparkFun MAX323 chip without the I2C IIC logic level converter (like in the screenshot from @commander-bln.
I use the code @irekzielinski 

Thank´s to all for you support!

[image]

closed 2023-08-25 by alexandergeiler