irekzielinski/Pylontech-Battery-Monitoring #24: Compiling with MQTT creates Errormessages

Vmie, 2023-08-06:
Hi all,
In general all works well and I've access to the Pylontech Batteries.
But if I like to enable the MQTT function, I get the following error, if I try to compile it:
....Documents\Arduino\libraries\ESP8266_Microgear\MicroGear.cpp: In member function 'void MicroGear::syncTime(Client*, long unsigned int*)':
....Documents\Arduino\libraries\ESP8266_Microgear\MicroGear.cpp:150:23: error: 'class WiFiClientSecure' has no member named 'setFingerprint'
         clientsecure->setFingerprint(tstr);
                       ^
....Documents\Arduino\libraries\ESP8266_Microgear\MicroGear.cpp:159:27: error: 'class WiFiClientSecure' has no member named 'setInsecure'
             clientsecure->setInsecure();

I couldn't fine any solution for this issue.
Maybe somebody know how to solve it.

Startertrek, 2023-10-18:
Actually I investet a lot of time to make this compiling, but no success.
My workaround is to reqest the data via the REST commands. This works very stable.

hidaba, 2023-12-23:
I think you use the incorrect library, try with [PubSubClient] library

TomtomWa, 2024-01-01:
thanks @hidaba, good hint, I had the same problem!

For anyone else using Arduino IDE (as I do), do the following:

- Go to library manager, search for "microgear" (ESP8266 Microgear) and click "Remove"
- Search for "PubSubClient" and install the library