irekzielinski/Pylontech-Battery-Monitoring #25: US3000C not respondig to the "pwr" string

Mirko-94-19, 2023-08-22:
I have 4 batteries, the master is US3000C and slave are mixed US2000C and US2000.
I have tried to use you sketch but the battery was not responding, then I connect directly to the console port using putty and it seems that the command to read the battery status is "bat" and not "pwr"

[image]

[image]

Not sure why it finds 14 batteries and why it differs from standard "pwr" command, maybe due to a new firmware

cschweers, 2023-08-23:
Your interpretation of the '_bat_' command is incorrect. This command shows the data for all **15** LiIon cells that make up the battery you are currently connected to.

Mirko-94-19, 2023-08-23:
Yes, you are right, thanks! 
I misunderstood the "help" and the serial output.
I tested one more time after a battery power cycle and now it is responding correctly to the "pwr"string but the ESP is not updating the data over MQTT, not sure why
Tested also with different MAX3232 chip but it still the same, over MQTT I get only "Available": "online"