irekzielinski/Pylontech-Battery-Monitoring #30: pwr output changed

alexandergeiler, 2024-06-07:
Hi,
I have a pylontech US2000B and a pylontech US3000A and I have been using @irekzielinski Pylontech-Battery-Monitoring tool since more than a year.
I am getting the data from pylontech and publish it to my moskitto MQTT broker and use it there.
Normally I switch off my battery at SOC < 20%, but last week my own program did not work and my US2000B battery went down to 0%.

Since then I have not been able to get any data from my US2000B battery.
The output has been changed:
US2000B:
[image]
US3000A
[image]

[image]
[image]

Sure, I still get data from US3000A, but the irekzielinski program cannot read the data from US2000B.

Has anyone else seen this? Does anyone know how to change the output format?

alexandergeiler, 2024-06-09:
I don´t know how, but my DIP configuration was wrong.
I changed all DIP switched to "OFF" and it is working again.
[image]

closed 2024-06-09 by alexandergeiler

Rustimation, 2024-11-15:
I just discovered your issue after I fixed the problem differently.
Strangely though, I have all DIP switches set to OFF and still get the "other" response to the pwr command - which in turn renders JSON and MQTT output unusable. I have therefore changed Ireneus' software a bit to reflext the new parameter positions.
Please see closed issue #31 "New Pylontech Firmware - New Data Structure - New Problems".
Rgds
Chris